nanpy/nanpy-firmware #11: Add build test and CI

ponty, 2015-02-10:


ponty, 2015-02-10:
@astagi  
Please enable TravisCI for the project and add the badge.

astagi, 2015-02-11:
@ponty done! but I got this error on travis

Using worker: worker-linux-7a054b10-2.bb.travis-ci.org:travis-linux-9
system_info
Build system information
Build language: c
Build image provisioning date and time
Wed Feb  4 18:22:50 UTC 2015
Operating System Details
Distributor ID:	Ubuntu
Description:	Ubuntu 12.04 LTS
Release:	12.04
Codename:	precise
Linux Version
2.6.32-042stab090.5
Cookbooks Version
23bb455 https://github.com/travis-ci/travis-cookbooks/tree/23bb455
GCC version
gcc (Ubuntu/Linaro 4.6.3-1ubuntu5) 4.6.3
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
LLVM version
clang version 3.4 (tags/RELEASE_34/final)
Target: x86_64-unknown-linux-gnu
Thread model: posix
Pre-installed Ruby versions
ruby-1.9.3-p551
Pre-installed Node.js versions
v0.10.36
Pre-installed Go versions
1.4.1
Redis version
redis-server 2.8.19
riak version
2.0.2
MongoDB version
MongoDB 2.4.12
CouchDB version
couchdb 1.6.1
Neo4j version
1.9.4
Cassandra version
2.0.9
RabbitMQ Version
3.4.3
ElasticSearch version
1.4.0
Installed Sphinx versions
2.0.10
2.1.9
2.2.6
Default Sphinx version
2.2.6
Installed Firefox version
firefox 31.0esr
PhantomJS version
1.9.8
ant -version
Apache Ant(TM) version 1.8.2 compiled on December 3 2011
mvn -version
Apache Maven 3.2.5 (12a6b3acb947671f09b81f49094c53f426d8cea1; 2014-12-14T17:29:23+00:00)
Maven home: /usr/local/maven
Java version: 1.7.0_76, vendor: Oracle Corporation
Java home: /usr/lib/jvm/java-7-oracle/jre
Default locale: en, platform encoding: UTF-8
OS name: "linux", version: "2.6.32-042stab090.5", arch: "amd64", family: "unix"
git.checkout
0.07s$ git clone --depth=50 --branch=master git://github.com/nanpy/nanpy-firmware.git nanpy/nanpy-firmware
Cloning into 'nanpy/nanpy-firmware'...
[... 64 lines cut ...]
Selecting previously unselected package arduino.
Unpacking arduino (from .../arduino_1%3a1.0+dfsg-9_all.deb) ...
Selecting previously unselected package extra-xdg-menus.
Unpacking extra-xdg-menus (from .../extra-xdg-menus_1.0-4_all.deb) ...
Processing triggers for man-db ...
Processing triggers for ccache ...
Updating symlinks in /usr/lib/ccache ...
Processing triggers for hicolor-icon-theme ...
Processing triggers for shared-mime-info ...
Setting up libftdi1 (0.19-4) ...
Setting up libjna-java (3.2.7-4) ...
Setting up librxtx-java (2.2pre2-10) ...
Setting up binutils-avr (2.20.1-2) ...
Setting up gcc-avr (1:4.5.3-3) ...
Setting up avrdude (5.11.1-1) ...
Setting up avr-libc (1:1.7.1-2) ...
Setting up arduino-core (1:1.0+dfsg-9) ...
Setting up arduino (1:1.0+dfsg-9) ...
Setting up extra-xdg-menus (1.0-4) ...
Processing triggers for libc-bin ...
ldconfig deferred processing now taking place
0.02s$ sudo rm /usr/share/arduino/libraries/Robot_Control/Wire.h
rm: cannot remove `/usr/share/arduino/libraries/Robot_Control/Wire.h': No such file or directory
The command "sudo rm /usr/share/arduino/libraries/Robot_Control/Wire.h" failed and exited with 1 during .
Your build has been stopped.

astagi, 2015-02-11:
I made little changes to travis.yml, anyway I still get an error in WireClass, looks like endTransmission doesn't accept any parameter (https://travis-ci.org/nanpy/nanpy-firmware)

   build/local/WireClass.cpp: In member function 'virtual void nanpy::WireClass::elaborate(nanpy::MethodDescriptor*)':
    build/local/WireClass.cpp:31:75: error: no matching function for call to 'TwoWire::requestFrom(int, int, int)'
    build/lib_00/Wire/Wire.h:53:13: note: candidates are: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)
    build/lib_00/Wire/Wire.h:54:13: note:                 uint8_t TwoWire::requestFrom(int, int)
    build/local/WireClass.cpp:39:51: error: no matching function for call to 'TwoWire::endTransmission(int)'
    build/lib_00/Wire/Wire.h:52:13: note: candidate is: uint8_t TwoWire::endTransmission()
    scons: *** [build/local/WireClass.o] Error 1

astagi, 2015-02-11:
Looks like endTransmission doesn't accept parameters

https://github.com/nanpy/nanpy-firmware/blob/master/Nanpy/WireClass.cpp#L39

ponty, 2015-02-11:
Passing! I updated Arduino to 1.0.5 in travis config. The Wire interface changed in 1.0.1 and TravisCi has only 1.0.

astagi, 2015-02-11:
That's awesome! :) closing...

closed 2015-02-11 by astagi