nanpy/nanpy-firmware #12: Uploading nanpy firmware for Arduino

KyawZinLatt, 2015-02-26:
I have problem with uploading firmware to Arduino. May be the way I try is wrong although I follow the instruction here https://github.com/nanpy/nanpy-firmware. Is there any other way to upload firmware to Arduino? I have tried the explanation here http://www.instructables.com/id/Arduino-Raspberry-Pi-Internet-Radio/?ALLSTEPS . It is really easy and working for me. But that version is version 0.7. Now I want to try for latest version. Can you show explain me the way to do it just simple way?

astagi, 2015-02-26:
Hi @KyawZinLatt , did you try this http://nanpy.github.io/? After 0.8 make and upload are entirely made using Arduino IDE

KyawZinLatt, 2015-02-26:
yeah I have tried that. To perform upload what else I need to do before uploading?

astagi, 2015-02-26:
You need to call ./config.sh and then set the libraries you want to use in
cfg.h.

KyawZinLatt, 2015-02-26:
This question may be quite easy for you but I dont know how to set the libraries to use in cfg.h. How to set that?

astagi, 2015-02-26:
Just turn 0 into 1 if you want to use a particular library :) start with a
simple example like blink... which Arduino board are you using?

KyawZinLatt, 2015-02-27:
Currently I am using Arduino Uno board
where to change that value?

astagi, 2015-02-27:
You have to call ./configure.sh and then edit Nanpy/cfg.h. Here you can
find a compressed version of the firmware with some settings enabled:
https://dl.dropboxusercontent.com/u/35837126/nanpy-firmware.zip

Please read http://nanpy.github.io/ carefully