nanpy/nanpy-firmware #15: Nanpy firmware error

Nebulae-52, 2015-03-14:
I am trying to install nanpy on an Arduino, and it gives me this error when I upload it:

aseClass.h: In member function 'void nanpy::ObjectsManager<T>::elaborate(nanpy::MethodDescriptor*) [with T = DHT]':
Nanpy.ino:105:   instantiated from here
BaseClass.h:27: note: neither the destructor nor the class-specific operator delete will be called, even if they are declared when the class is defined.
BaseClass.h: In member function 'void nanpy::ObjectsManager<T>::elaborate(nanpy::MethodDescriptor*) [with T = CapacitiveSensor]':
Nanpy.ino:105:   instantiated from here
BaseClass.h:27: note: neither the destructor nor the class-specific operator delete will be called, even if they are declared when the class is defined.
BaseClass.h: In member function 'void nanpy::ObjectsManager<T>::elaborate(nanpy::MethodDescriptor*) [with T = LiquidCrystal_I2C]':
Nanpy.ino:105:   instantiated from here
BaseClass.h:27: note: neither the destructor nor the class-specific operator delete will be called, even if they are declared when the class is defined.
BaseClass.h: In member function 'void nanpy::ObjectsManager<T>::elaborate(nanpy::MethodDescriptor*) [with T = LiquidCrystal]':
Nanpy.ino:105:   instantiated from here
BaseClass.h:27: note: neither the destructor nor the class-specific operator delete will be called, even if they are declared when the class is defined.
BaseClass.h: In member function 'void nanpy::ObjectsManager<T>::elaborate(nanpy::MethodDescriptor*) [with T = DallasTemperature]':
Nanpy.ino:105:   instantiated from here
BaseClass.h:27: note: neither the destructor nor the class-specific operator delete will be called, even if they are declared when the class is defined.
BaseClass.h: In member function 'void nanpy::ObjectsManager<T>::elaborate(nanpy::MethodDescriptor*) [with T = Servo]':
Nanpy.ino:105:   instantiated from here
BaseClass.h:27: note: neither the destructor nor the class-specific operator delete will be called, even if they are declared when the class is defined.
BaseClass.h: In member function 'void nanpy::ObjectsManager<T>::elaborate(nanpy::MethodDescriptor*) [with T = Stepper]':
Nanpy.ino:105:   instantiated from here
BaseClass.h:27: note: neither the destructor nor the class-specific operator delete will be called, even if they are declared when the class is defined.
BaseClass.h: In member function 'void nanpy::ObjectsManager<T>::elaborate(nanpy::MethodDescriptor*) [with T = OneWire]':
Nanpy.ino:105:   instantiated from here
BaseClass.h:27: note: neither the destructor nor the class-specific operator delete will be called, even if they are declared when the class is defined.

The code uploads, but when I run any code, even the examples, the led blinks twice and then it gives me a serial timeout: 

$python blink.py 
Traceback (most recent call last):
  File "blink.py", line 11, in <module>
    a = ArduinoApi(connection=connection)
  File "build/bdist.macosx-10.9-intel/egg/nanpy/classinfo.py", line 22, in getinstance
  File "build/bdist.macosx-10.9-intel/egg/nanpy/classinfo.py", line 59, in __init__
  File "build/bdist.macosx-10.9-intel/egg/nanpy/memo.py", line 20, in __call__
  File "build/bdist.macosx-10.9-intel/egg/nanpy/arduinoboard.py", line 57, in wrapper
  File "build/bdist.macosx-10.9-intel/egg/nanpy/arduinoboard.py", line 125, in wrapper
  File "build/bdist.macosx-10.9-intel/egg/nanpy/arduinoboard.py", line 47, in _call
  File "build/bdist.macosx-10.9-intel/egg/nanpy/arduinoboard.py", line 16, in return_value
  File "build/bdist.macosx-10.9-intel/egg/nanpy/serialmanager.py", line 98, in readline
nanpy.serialmanager.SerialManagerError: Serial timeout!

running on a macbook

Nebulae-52, 2015-03-15:
I got the blink example to work on a different computer, but Arduino gave me the same error when the firmware was uploading

astagi, 2015-03-20:
Have you edit the cfg.h file?

astagi, 2016-02-06:
@Nebulae-52 any update on this?

closed 2016-03-10 by astagi