nanpy/nanpy-firmware #20: Possible firmware issue

Lichthof, 2016-02-03:
I'm by no means an expert in the programming field, I've just started with an Arduino getting a speech synth chip to work. When trying to load some code that imports nanpy I get the following readout

E:\python>python serial.py
Traceback (most recent call last):
  File "serial.py", line 6, in <module>
    from nanpy import (arduinoapi, serial_manager)
  File "E:\Python 3.5.1\lib\site-packages\nanpy-0.9.5-py3.5.egg\nanpy\__init__.py", line 12, in <module>
  File "E:\Python 3.5.1\lib\site-packages\nanpy-0.9.5-py3.5.egg\nanpy\arduinoboard.py", line 1, in <module>
  File "E:\Python 3.5.1\lib\site-packages\nanpy-0.9.5-py3.5.egg\nanpy\serialmanager.py", line 3, in <module>
  File "E:\python\serial.py", line 6, in <module>
    from nanpy import (arduinoapi, serial_manager)
  File "E:\Python 3.5.1\lib\site-packages\nanpy-0.9.5-py3.5.egg\nanpy\arduinoapi.py", line 1, in <module>
ImportError: cannot import name 'arduinomethod'

The error is something to do with the arduinomethod(???) and I have no idea how I can fix the problem. Any help would be much appreciated.

astagi, 2016-02-06:
Looks like a problem with the python part. Nanpy-firmware must be flashed in the arduino board, Nanpy python part must be installed in your system or your virtual environment (more info here: http://nanpy.github.io/)

Try to install both of them and run some python example from here: https://github.com/nanpy/nanpy/tree/master/nanpy/examples

Lichthof, 2016-02-06:
I had already uploaded the firmware to the Arduino, and nanpy is installed on my PC. I've reinstalled them both and I'm still getting the same error. The examples create the same error too. It's probably something I'm doing wrong when installing something, but I can't see what it is that I'm doing wrong.

astagi, 2016-02-06:
Ok! Could you please show me the code of serial.py? I suppose this is a script you wrote and that is using Nanpy

Lichthof, 2016-02-07:
Sorry about the delay. This is the code I am using. http://pastebin.com/zH0AGC92

astagi, 2016-02-07:
No problem @Lichthof . The problem is that you're using old api, try with this example: http://pastebin.com/7haT2mXX

It should work! For more info: http://nanpy.github.io/

Lichthof, 2016-02-07:
I've tried that and now I'm getting this :(

Traceback (most recent call last):
  File "E:\python\serial.py", line 6, in <module>
    from nanpy import ArduinoApi, SerialManager
  File "E:\Python 3.5.1\lib\site-packages\nanpy\__init__.py", line 12, in <module>
    from nanpy.arduinoboard import ArduinoObject
  File "E:\Python 3.5.1\lib\site-packages\nanpy\arduinoboard.py", line 1, in <module>
    from nanpy.serialmanager import serial_manager
  File "E:\Python 3.5.1\lib\site-packages\nanpy\serialmanager.py", line 3, in <module>
    import serial
  File "E:\python\serial.py", line 6, in <module>
    from nanpy import ArduinoApi, SerialManager
ImportError: cannot import name 'ArduinoApi'
>>>

astagi, 2016-02-07:
Do you use Nanpy 0.9.5?

Lichthof, 2016-02-07:
Yes. I reinstalled twice using pip and easy_install, and both still create the same error.

astagi, 2016-02-08:
uhm really  weird.. please @Lichthof could you execute

    pip freeze | grep nanpy

and paste here the output?

Lichthof, 2016-02-08:
Here's the output:

e:\python>pip freeze Grep nanpy
nanpy==0.9.5
nltk==3.1
pyserial==3.0.1

Lichthof, 2016-02-10:
I've tried installing it all on my Surface Pro, and I'm still getting the same errors. :(

ponty, 2016-02-11:
try this:

```python
import nanpy
from pprint import pprint
pprint(nanpy.__dict__)
```

Lichthof, 2016-02-11:
This is the output - http://pastebin.com/2T0SNZyp

ponty, 2016-02-12:
Rename your script!

It has the same name as the `serial` module.

Lichthof, 2016-02-12:
Brilliant, thank you very much! But from one problem to another I'm now getting this :(

  File "E:\python\ssi.py", line 15, in <module>
    a1 = arduinoapi(connection=connection_1)
TypeError: 'module' object is not callable

astagi, 2016-02-12:
great catch @ponty :smile: 

@Lichthof please follow this example to use imported modules correctly https://github.com/nanpy/nanpy/blob/master/nanpy/examples/blink_2boards.py (or one of these example in the source https://github.com/nanpy/nanpy/blob/master/nanpy/examples/)

astagi, 2016-02-13:
@Lichthof can we close this issue?

Lichthof, 2016-02-13:
Yes! Thank you very much for your help and sorry it was such a silly problem!

astagi, 2016-02-13:
No problem @Lichthof :smile:

closed 2016-02-13 by astagi